chipsalliance/riscv-vector-tests #12: Need more test case for vector index load/store

huxuan0307, 2023-09-05:
The vector extension allows index load insts to use different width of data and index. For example:

```asm
vsetvli t1, t0, e8,m1,ta,ma
vluxei64.v v1, (a0), v2

vsetvli t1, t0, e16,m1,ta,ma
vluxseg2ei64.v v2, (a0), v4
```

ksco, 2023-09-27:
Added support for different data/offset widths in the above commit, sorry for the delay. The generated tests look good for me now, feel free to reopen this if there is any follow-up issue.

jerryz123, 2023-09-27:
@ksco That patch results in code that causes accesses to undefined/out-of-bounds regions.
I do `make VLEN=256 SPLIT=2000 MODE=virtual`, which results in two types of failures:
- Test failures in stage2 (but not in stage1)... this is because some tests now generate indexed accesses off the end of the `testdata` array. That region can change between `stage1` and `stage2`
- Page handling errors ... this is likely related to the above... this happens because there is a small bound on the size of the paged-in region for virtual-env tests.

IMO, the easiest way to resolve this is to fold in the index-generation and indexed-data generation into a single step, so the generator can guarantee that all indexed accesses like within the generated region. 

Segmented loads/stores make this more complex as well.

jerryz123, 2023-09-28:
Oops I forgot, my make command only works with latest riscv-test-env, that should be bumped

ksco, 2023-09-28:
Ok, let me bump it and try.

ksco, 2023-09-28:
Fixed, the main issue is we are using the same data section for all split tests, while it's okay for non-virtual env, I think it's sometimes too far for the virtual env. Anyway, I split the data section in above commit, and it works.

ksco, 2023-09-28:
Currently it's only a workaround, the generator code is ugly and certainly needs some refactoring.

closed 2024-04-07 by ksco