chipsalliance/riscv-vector-tests #15: How to use this project for plct-gem5？

luojihe, 2023-11-16:
Hello！I saw from the weekly report of PLCT Lab that this project is for testing plclt gem5.
I took some time to compile it, but there were errors when running it on plclt gem5. 
Can you give us a demonstration example of how to use it?

My command on plclt gem5
`../build/RISCV/gem5.opt  ../configs/example/se.py --cpu-type RiscvAtomicSimpleCPU --cmd v256x64machine/bin/stage1/vadd.vx-0`

Error info：
```
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version [DEVELOP-FOR-22.1]
gem5 compiled Nov  8 2023 14:16:47
gem5 started Nov 16 2023 14:25:31
gem5 executing on ic610.hochip.cc, pid 31407
command line: ../build/RISCV/gem5.opt ../configs/example/se.py --cpu-type RiscvAtomicSimpleCPU --cmd v256x64machine/bin/stage1/vadd.vx-0

Global frequency set at 1000000000000 ticks per second
build/RISCV/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (512 Mbytes)
build/RISCV/arch/riscv/linux/se_workload.cc:60: warn: Unknown operating system; assuming Linux.
0: system.remote_gdb: listening for remote gdb on port 7000
**** REAL SIMULATION ****
build/RISCV/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
build/RISCV/arch/riscv/faults.cc:196: panic: Illegal instruction 0x22050f1402573 at pc (0x80000086=>0x8000008a).(0=>1): mhartid is not accessible in 0
Memory Usage: 793476 KBytes
Program aborted at tick 16500
--- BEGIN LIBC BACKTRACE ---
../build/RISCV/gem5.opt[0x17425f5]
../build/RISCV/gem5.opt[0x17633e9]
/lib64/libpthread.so.0(+0xf370)[0x2b502bb05370]
/lib64/libc.so.6(gsignal+0x37)[0x2b502cc781d7]
/lib64/libc.so.6(abort+0x148)[0x2b502cc798c8]
../build/RISCV/gem5.opt[0x7c39bf]
../build/RISCV/gem5.opt[0x7c286d]
../build/RISCV/gem5.opt[0x7c2a62]
../build/RISCV/gem5.opt[0xf25fd7]
../build/RISCV/gem5.opt[0xf0adae]
../build/RISCV/gem5.opt[0x1754c8a]
../build/RISCV/gem5.opt[0x1779809]
../build/RISCV/gem5.opt[0x177a4ea]
../build/RISCV/gem5.opt[0x16e991e]
../build/RISCV/gem5.opt[0x7a2b08]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(+0x986b8)[0x2b502b6126b8]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x7c4f)[0x2b502b5ee85f]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x8df)[0x2b502b6eecef]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(_PyFunction_Vectorcall+0x91)[0x2b502b612a81]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(+0x6c7fd)[0x2b502b5e67fd]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x67ed)[0x2b502b5ed3fd]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(+0x6b533)[0x2b502b5e5533]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(+0x6c7fd)[0x2b502b5e67fd]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x2af3)[0x2b502b5e9703]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x8df)[0x2b502b6eecef]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(_PyFunction_Vectorcall+0x91)[0x2b502b612a81]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(+0x6c7fd)[0x2b502b5e67fd]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x67ed)[0x2b502b5ed3fd]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x8df)[0x2b502b6eecef]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8..so.1.0(PyEval_EvalCodeEx+0x3e)[0x2b502b6ef08e]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(PyEval_EvalCode+0x1b)[0x2b502b6ef0bb]
/eda/python/python3.8.13_gcc10.2.0/lib/libpython3.8.so.1.0(+0x171a71)[0x2b502b6eba71]
--- END LIBC BACKTRACE ---
Aborted (core dumped)
```

Thanks！

ksco, 2023-11-16:
You need to use `-e MODE=user` to build userland tests, please read `How to use` section in README.

luojihe, 2023-11-16:
> You need to use `-e MODE=user` to build userland tests, please read `How to use` section in README.

Thanks. 
I am a novice in this field and I still have some questions to consult.
After following the above suggestions, I encountered other issues.

When I change the mode to ‘user’, I get stuck in the tests_patch step, such as
`LD_LIBRARY_PATH=/opt/riscv64/lib build/pspike --isa=rv64gcv --varch=vlen:256,elen:64 out/v256x64user/bin/stage1/vaadd.vv-0 > out/v256x64user/patches/stage2/vaadd.vv-0.patch`
The program didn't respond for hours when it arrived here.
(Later, I tried the machine mode and it was okay)

If I directly use the bin file of stage1, gem5 will also report an error
`../build/RISCV/gem5.opt  ../configs/example/se.py --cpu-type RiscvAtomicSimpleCPU --cmd v256x64user/bin/stage1/vadd.vi-0`
`build/RISCV/arch/riscv/faults.cc:188: panic: Unknown instruction 0x0000800b at pc (0x80000072=>0x80000076).(0=>1)`

ksco, 2023-11-16:
Oh, user mode support is actually broken and I've fixed it above, can you pull and try again?

luojihe, 2023-11-17:
Thanks very much! It works.
I can now run the generated binaries file on plct-gem5.

ksco, 2023-11-17:
Great!

closed 2023-11-17 by ksco