chipsalliance/riscv-vector-tests #18: Make for target 'build-patcher-spike' failed

ptzheng, 2023-12-07:
After meeting prerequisites, I execute `make -j40`. However, I got the error message as follows：
[image]
I check the related files and find that the function `cfg_t::cfg` is called in `pspike/psike.cc` with 12 arguments provided:
[image]
But the candidate function in my pre-built **Spike** expects 13  arguments:
[image]
It seems that the call for `cfg_t::cfg` in `pspike/psike.cc` missing the `const char * default_priv` argument. Could you please help me solve the problem?

ksco, 2023-12-07:
We're using some unstable API, so it changes from time to time, please check [build-and-test.yml](https://github.com/ksco/riscv-vector-tests/blob/main/.github/workflows/build-and-test.yml#L47) file to use a specific version of spike.

ptzheng, 2023-12-07:
> We're using some unstable API, so it changes from time to time, please check [build-and-test.yml](https://github.com/ksco/riscv-vector-tests/blob/main/.github/workflows/build-and-test.yml#L47) file to use a specific version of spike.

Thanks, it works. 
However, When I change the mode to ‘user’, I get stuck in the `tests_patch step`, such as
`LD_LIBRARY_PATH=/riscv_toolchains/lib build/pspike --isa=rv64gcv --varch=vlen:256,elen:64 /riscv_toolchains/riscv64-unknown-elf/bin/pk out/v256x64user/bin/stage1/vaadd.vv-0 > out/v256x64user/patches/stage2/vaadd.vv-0.patch`
I get the error message:
`*** FAILED *** (tohost = -1073707433)`
`*** FAILED *** (tohost = -1073707433)`

ksco, 2023-12-07:
That's strange. `out/v256x64user/bin/stage1/vaadd.vv-0` is not supposed to have this many test cases. The last few lines on my side are:

```
  TEST_CASE(2111, t0, 0xffffffffffffc000, ld t0, 0(a0); addi a0, a0, 8)
  TEST_CASE(2112, t0, 0xa989e5f32751c914, ld t0, 0(a0); addi a0, a0, 8)
  TEST_CASE(2113, t0, 0xffffffffbfffc000, ld t0, 0(a0); addi a0, a0, 8)
---
```

ksco, 2023-12-07:
Are you sure you're using the latest `main` without any changes?

ksco, 2023-12-07:
Then I don't know why, the "patch" stage works fine on my side. Spike with pk is indeed a bit unstable for reasons I don't know, but the errors only happen at the later "merger" stage on my side, retrying a few times will get a successful build.

ksco, 2024-12-20:
Closed for long periods of inactivity.

closed 2024-12-20 by ksco