chipsalliance/riscv-vector-tests #21: Add more tests which check CSR functionality

troibe, 2024-01-03:
Thanks for the test suite. It's super helpful for developing simulators and RTL.
Maybe I missed something but I think the tests currently don't take into accout `vxrm`, `frm`, `mstatus`, `sstatus`, `vsstatus` and other CSRs.

ksco, 2024-01-04:
I'm currently busy with other things, but I will add them.

ksco, 2024-01-09:
I already added vxrm and vxsat support.

troibe, 2024-01-09:
Thanks so much! :tada:

troibe, 2024-01-29:
I think some `vstart` related tests would also be helpful.
So far my ISA simulator always set `vstart` to zero which should lead to issues I think
but there are no test cases covering it as far as I can tell.

ksco, 2024-01-30:
Normally, `vstart` is written by hardware on a trap, which is not straightforward to fit into our test framework. `vstart` can also be written by `csrrw[i]`, which is a usage scenario not recommended by the v-spec, I can at least add support for the latter case.

jerryz123, 2024-01-30:
Note that implementations are not required to support non-zero vstart for instructions where they would never take a trap in the middle of execution.

ksco, 2024-01-30:
Indeed. Then we have to make `vstart` support optional for arithmetic instructions if we add it to the test.

Page 14: 
> For example, some implementations will never take interrupts during execution of a vector arithmetic instruction, instead waiting until the instruction completes to take the interrupt. Such implementations are permitted to raise an illegal instruction exception when attempting to execute a vector arithmetic instruction when `vstart` is nonzero.

ksco, 2024-01-30:
It is also possible that an implementation that does not support exceptions only supports `vstart == 0`.