chipsalliance/riscv-vector-tests #37: for VLEN=2048 test script gives errors.

akifejaz, 2024-03-26:
I've virtual platform which supports the VLEN=2048b, but when I tried to generate tests for 2048, I'm getting issue with some instructions like vmv8r. Can you please help me with this?

`/tools/rvv-intrinsic-ci6.1/bin/../lib/gcc/riscv64-unknown-elf/13.0.1/../../../../riscv64-unknown-elf/bin/ld: warning: out/bin/stage1/vmv.v.x-0 has a LOAD segment with RWX permissions
/tools/rvv-intrinsic-ci6.1/bin/../lib/gcc/riscv64-unknown-elf/13.0.1/../../../../riscv64-unknown-elf/bin/ld: warning: riscv64-unknown-elf-gcc -march=rv64gcv -mabi=lp64d -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Ienv/p -Imacros/general -Tenv/p/link.ld out/tests/stage1/vmxor.mm-0.S -o out/bin/stage1/vmxor.mm-0
out/bin/stage1/vmv.v.v-0 has a LOAD segment with RWX permissions
out/tests/stage1/vmv8r.v-0.S: Assembler messages:
out/tests/stage1/vmv8r.v-0.S:25: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:60: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:95: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:130: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:165: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:200: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:235: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:270: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:305: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:340: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:375: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:410: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:445: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:480: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:515: Error: illegal operands `addi a0,a0,2048'
out/tests/stage1/vmv8r.v-0.S:550: Error: illegal operands `addi a0,a0,2048'
/tools/rvv-intrinsic-ci6.1/bin/../lib/gcc/riscv64-unknown-elf/13.0.1/../../../../riscv64-unknown-elf/bin/ld: warning: out/bin/stage1/vmv1r.v-0 has a LOAD segment with RWX permissions
make[1]: *** [Makefile:65: vmv8r.v-0] Error 1
make[1]: *** Waiting for unfinished jobs....
/tools/rvv-intrinsic-ci6.1/bin/../lib/gcc/riscv64-unknown-elf/13.0.1/../../../../riscv64-unknown-elf/bin/ld: warning: out/bin/stage1/vmv.x.s-0 has a LOAD segment with RWX permissions
/tools/rvv-intrinsic-ci6.1/bin/../lib/gcc/riscv64-unknown-elf/13.0.1/../../../../riscv64-unknown-elf/bin/ld: warning: out/bin/stage1/vmv2r.v-0 has a LOAD segment with RWX permissions
/tools/rvv-intrinsic-ci6.1/bin/../lib/gcc/riscv64-unknown-elf/13.0.1/../../../../riscv64-unknown-elf/bin/ld: warning: out/bin/stage1/vmv4r.v-0 has a LOAD segment with RWX permissions
/tools/rvv-intrinsic-ci6.1/bin/../lib/gcc/riscv64-unknown-elf/13.0.1/../../../../riscv64-unknown-elf/bin/ld: warning: out/bin/stage1/vmxnor.mm-0 has a LOAD segment with RWX permissions
/tools/rvv-intrinsic-ci6.1/bin/../lib/gcc/riscv64-unknown-elf/13.0.1/../../../../riscv64-unknown-elf/bin/ld: warning: out/bin/stage1/vmxor.mm-0 has a LOAD segment with RWX permissions
make[1]: Leaving directory '/projects/hydra/t_aejaz/rv-tests'
make: *** [Makefile:62: compile-stage1] Error 2`

Best,
Akif

@ksco

ksco, 2024-03-26:
Please run `git pull` to use the latest version.

akifejaz, 2024-03-29:
> Please run `git pull` to use the latest version.

I've pull the latest changes, I'm encountering some issue that i didn't faced in previous version.  Can you please help me with this what can be the issue?

`t_aejaz@rhel-compute-2:/projects/hydra/t_aejaz/rv-tests$ make -j$(nproc)`
`rm -rf build
rm -rf out/v256x64machine
go build -o build/generator 
mkdir -p build
go build -o build/merger merger/merger.go
g++ -std=c++17 -I../build/spike/ -I../build/spike/build -I../build/spike/softfloat -I../build/spike/fesvr -L../build/spike/build pspike/pspike.cc -lriscv -lfesvr -o build/pspike
pspike/pspike.cc:10:37: error: ‘abstract_mem_t’ was not declared in this scope; did you mean ‘abstractauto_t’?
   10 | static std::vector<std::pair<reg_t, abstract_mem_t*>> make_mems(const std::vector<mem_cfg_t> &layout)
      |                                     ^~~~~~~~~~~~~~
      |                                     abstractauto_t
pspike/pspike.cc:10:51: error: template argument 2 is invalid
   10 | static std::vector<std::pair<reg_t, abstract_mem_t*>> make_mems(const std::vector<mem_cfg_t> &layout)
      |                                                   ^
pspike/pspike.cc:10:52: error: template argument 1 is invalid
   10 | static std::vector<std::pair<reg_t, abstract_mem_t*>> make_mems(const std::vector<mem_cfg_t> &layout)
      |                                                    ^~
pspike/pspike.cc:10:52: error: template argument 2 is invalid
pspike/pspike.cc: In function ‘int make_mems(const std::vector<mem_cfg_t>&)’:
pspike/pspike.cc:12:32: error: ‘abstract_mem_t’ was not declared in this scope; did you mean ‘abstractauto_t’?
   12 |   std::vector<std::pair<reg_t, abstract_mem_t*>> mems;
      |                                ^~~~~~~~~~~~~~
      |                                abstractauto_t
pspike/pspike.cc:12:46: error: template argument 2 is invalid`

ksco, 2024-03-29:
Refer to [build-and-test.yml](https://github.com/ksco/riscv-vector-tests/blob/main/.github/workflows/build-and-test.yml) for instructions to build the project. Basically, you need to stick the Spike to a specific version as we are using some unstable API.

akifejaz, 2024-04-01:
Hi, For my Core I've to change few linker addresses but when I'm trying to give patched spike that address with option -m it say it is unrecognized. Is there any other way to set the spike start address?

```make[1]: Entering directory '/projects/hydra/t_aejaz/rv-tests'
LD_LIBRARY_PATH=spike/riscv-isa-sim/build build/pspike --isa=rv64gcv --varch=vlen:2048,elen:64  -m0x800000000:0x1000000 out/v2048x64machine/bin/stage1/vl1re16.v-0 > out/v2048x64machine/patches/stage2/vl1re16.v-0.patch
LD_LIBRARY_PATH=spike/riscv-isa-sim/build build/pspike --isa=rv64gcv --varch=vlen:2048,elen:64  -m0x800000000:0x1000000 out/v2048x64machine/bin/stage1/vl1re32.v-0 > out/v2048x64machine/patches/stage2/vl1re32.v-0.patch
LD_LIBRARY_PATH=spike/riscv-isa-sim/build build/pspike --isa=rv64gcv --varch=vlen:2048,elen:64  -m0x800000000:0x1000000 out/v2048x64machine/bin/stage1/vl1re64.v-0 > out/v2048x64machine/patches/stage2/vl1re64.v-0.patch
LD_LIBRARY_PATH=spike/riscv-isa-sim/build build/pspike --isa=rv64gcv --varch=vlen:2048,elen:64  -m0x800000000:0x1000000 out/v2048x64machine/bin/stage1/vl1re8.v-0 > out/v2048x64machine/patches/stage2/vl1re8.v-0.patch
build/pspike: unrecognized option -m0x800000000:0x1000000
make[1]: *** [Makefile:95: vl1re16.v-0.patch] Error 1
make[1]: *** Waiting for unfinished jobs....
build/pspike: unrecognized option -m0x800000000:0x1000000
make[1]: *** [Makefile:95: vl1re32.v-0.patch] Error 1
build/pspike: unrecognized option -m0x800000000:0x1000000
make[1]: *** [Makefile:95: vl1re64.v-0.patch] Error 1
build/pspike: unrecognized option -m0x800000000:0x1000000
make[1]: *** [Makefile:95: vl1re8.v-0.patch] Error 1
make[1]: Leaving directory '/projects/hydra/t_aejaz/rv-tests'
make: *** [Makefile:92: patching-stage2] Error 2
```

akifejaz, 2024-04-02:
Hi @ksco , Any thoughts on this above problem?

akifejaz, 2024-04-02:
one update from myside, --I tried to build the latest spike of commit mentioned in [build-and-test.yml](https://github.com/ksco/riscv-vector-tests/blob/main/.github/workflows/build-and-test.yml) , and when I tried to directly run on it, the tests seems to fail with tohost value 668. 

run cmd:
```
make -e VLEN=2048 XLEN=64 MODE=machine PATTERN='^v[l].+\.v$' -j$(nproc)
```

```
tools/experimental/riscv-isa-sim-commitlog/bin/spike --isa=rv64gcv --varch=vlen:2048,elen:64 -m0x800000000:0x100000000 out/v2048x64machine/bin/stage1/vl2re16.v-0 > out/v2048x64machine/patches/stage2/vl2re16.v-0.patch
/tools/experimental/riscv-isa-sim-commitlog/bin/spike --isa=rv64gcv --varch=vlen:2048,elen:64 -m0x800000000:0x100000000 out/v2048x64machine/bin/stage1/vl2re32.v-0 > out/v2048x64machine/patches/stage2/vl2re32.v-0.patch
/tools/experimental/riscv-isa-sim-commitlog/bin/spike --isa=rv64gcv --varch=vlen:2048,elen:64 -m0x800000000:0x100000000 out/v2048x64machine/bin/stage1/vl2re64.v-0 > out/v2048x64machine/patches/stage2/vl2re64.v-0.patch
/tools/experimental/riscv-isa-sim-commitlog/bin/spike --isa=rv64gcv --varch=vlen:2048,elen:64 -m0x800000000:0x100000000 out/v2048x64machine/bin/stage1/vl2re8.v-0 > out/v2048x64machine/patches/stage2/vl2re8.v-0.patch
*** FAILED *** (tohost = 668)
*** FAILED *** (tohost = 668)
make[2]: *** [Makefile:96: vl1re32.v-0.patch] Error 156
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:96: vl2re32.v-0.patch] Error 156
*** FAILED *** (tohost = 668)
*** FAILED *** (tohost = 668)
*** FAILED *** (tohost = 668)
*** FAILED *** (tohost = 668)
make[2]: *** [Makefile:96: vl1re16.v-0.patch] Error 156
make[2]: *** [Makefile:96: vl1re8.v-0.patch] Error 156
make[2]: *** [Makefile:96: vl2re8.v-0.patch] Error 156
make[2]: *** [Makefile:96: vl2re16.v-0.patch] Error 156
*** FAILED *** (tohost = 668)
```

ksco, 2024-04-02:
Because we didn't support the -m option in patched spike. Take a look at [pspike.cc:59](https://github.com/ksco/riscv-vector-tests/blob/main/pspike/pspike.cc#L59). And for the 668 error, stage 1 tests can only run under patched spike, we patched it for a reason.

akifejaz, 2024-04-02:
I've changed the memory addr, but I think there are more places where I need to mention my new address?? -- Actually the problem the I've added new linker which has different address space like it starts with 0x8_0000_0000. 

This is the linker I'm using to generate tests. 
```
IMEM_START       = 0x800000000;

IMEM_BASE        = IMEM_START;
IMEM_END         = IMEM_START + (768*1024);
STACK_SIZE       = 16K;
STACK_FENCE_SIZE = 16;
CMEM_BASE        = 0x100000000;

SECTIONS
{
    . = IMEM_BASE;
    .boot   : { *(.boot) }
    .text : { *(.text) }
    .data : { *(.data) }
    .bss : { *(.bss) }
    . = ALIGN(8);
    .tohost : { *(.tohost) }
    . = ALIGN(8);
    STACK_END = .;
    .stack_fence : { */xc_boot.o(._stack_fence) }
    . = STACK_END + STACK_SIZE + STACK_FENCE_SIZE;
    STACK_BASE = .;
    . = ALIGN(8);
    _end = .;
}
```

The error after updating the memory addr, on [pspike.cc:59](https://github.com/ksco/riscv-vector-tests/blob/main/pspike/pspike.cc#L59). 

```
make[1]: Entering directory '/projects/hydra/t_aejaz/rv-tests'
LD_LIBRARY_PATH=spike/riscv-isa-sim/build build/pspike --isa=rv64gcv --varch=vlen:2048,elen:64  out/v2048x64machine/bin/stage1/vl1re16.v-0 > out/v2048x64machine/patches/stage2/vl1re16.v-0.patch
LD_LIBRARY_PATH=spike/riscv-isa-sim/build build/pspike --isa=rv64gcv --varch=vlen:2048,elen:64  out/v2048x64machine/bin/stage1/vl1re32.v-0 > out/v2048x64machine/patches/stage2/vl1re32.v-0.patch
LD_LIBRARY_PATH=spike/riscv-isa-sim/build build/pspike --isa=rv64gcv --varch=vlen:2048,elen:64  out/v2048x64machine/bin/stage1/vl1re64.v-0 > out/v2048x64machine/patches/stage2/vl1re64.v-0.patch
LD_LIBRARY_PATH=spike/riscv-isa-sim/build build/pspike --isa=rv64gcv --varch=vlen:2048,elen:64  out/v2048x64machine/bin/stage1/vl1re8.v-0 > out/v2048x64machine/patches/stage2/vl1re8.v-0.patch
Memory address 0x800000000 is invalid
Access exception occurred while loading payload out/v2048x64machine/bin/stage1/vl1re8.v-0:
Memory address 0x800000000 is invalid
LD_LIBRARY_PATH=spike/riscv-isa-sim/build build/pspike --isa=rv64gcv --varch=vlen:2048,elen:64  out/v2048x64machine/bin/stage1/vl2re8.v-0 > out/v2048x64machine/patches/stage2/vl2re8.v-0.patch
make[1]: *** [Makefile:95: vl1re64.v-0.patch] Error 255
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:95: vl1re8.v-0.patch] Error 255
Access exception occurred while loading payload out/v2048x64machine/bin/stage1/vl2re16.v-0:
Memory address 0x800000000 is invalid
make[1]: *** [Makefile:95: vl2re16.v-0.patch] Error 255
Access exception occurred while loading payload out/v2048x64machine/bin/stage1/vl1re32.v-0:
Memory address 0x800000000 is invalid
make[1]: *** [Makefile:95: vl1re32.v-0.patch] Error 255
Access exception occurred while loading payload out/v2048x64machine/bin/stage1/vl1re16.v-0:
Memory address 0x800000000 is invalid
```

akifejaz, 2024-04-03:
Hi @ksco, sorry to disturb but any thoughts on the above problem?

ksco, 2024-04-03:
I pushed a fix bcbda657a2cdd1b91ea0d52f9893fcce89a6bbed on pspike.cc.

akifejaz, 2024-04-06:
> I pushed a fix [bcbda65](https://github.com/ksco/riscv-vector-tests/commit/bcbda657a2cdd1b91ea0d52f9893fcce89a6bbed) on pspike.cc.

Hi Thanks for your help, It solved my problem. I'm able to generate the vlen2048 tests with custom address space now.

ksco, 2024-04-06:
Okay, then I'll close this issue.

closed 2024-04-06 by ksco