chipsalliance/riscv-vector-tests #40: VLEN 2048 tests size

FariaaFaheem, 2024-04-24:
I've virtual platform which supports the VLEN=2048b, when I tried to generate tests for 2048. The size of the executable files is way too large. Is there something we can configure to make it smaller?

ksco, 2024-04-24:
Yes, there is an undocumented option called SPLIT, see [Makefile#L4](https://github.com/ksco/riscv-vector-tests/blob/main/Makefile#L4), you can try to make it smaller to get a smaller per file size. I'll document that option later.

akifejaz, 2024-04-24:
> Yes, there is an undocumented option called SPLIT, see [Makefile#L4](https://github.com/ksco/riscv-vector-tests/blob/main/Makefile#L4), you can try to make it smaller to get a smaller per file size. I'll document that option later.

in general, is this the number of lines of code? or is it something else?

ksco, 2024-05-16:
I've added documentation for all available option, use `make help` to see all of them.

closed 2024-05-16 by ksco