chipsalliance/riscv-vector-tests #57: How to check resultdata

chenyuanyping, 2024-12-17:
Case: out/v128x32machine/tests/stage1/vaadd_vv-0.S
It seems that the final results were not checked and compared.
[image]

ksco, 2024-12-17:
The final assembly is in stage2, not stage1.

chenyuanyping, 2024-12-17:
Thanks for the answer. I also want to know the function of the .word command.
And When I generated stage2, the following error occurred
[image]

ksco, 2024-12-18:
Can you try XLEN=64 to see if it's working? Maybe something is wrong under XLEN=32?

chenyuanyping, 2024-12-18:
I have tried and the error occurs as follows
[image]

ksco, 2024-12-18:
Maybe something is wrong with your setup, check how we did this in the CI, refer to https://github.com/chipsalliance/riscv-vector-tests/blob/main/.github/workflows/build-and-test.yml.

chenyuanyping, 2024-12-18:
Thanks,I will try

ksco, 2024-12-18:
Oh, about the magic .word, check the "How it works" section in README. We modified spike to add a "magic" instruction which generates reference result checking code, and then merged back to the original assembly file.

chenyuanyping, 2024-12-18:
thanks for modified.  I tried VLEN=128,XLEN=64 and it passed, but when I tried VLEN=128,XLEN=32, it gave the error as below
[image]

chenyuanyping, 2024-12-18:
I run 
 `make generate-stage1 --environment-overrides VLEN=128 XLEN=32 ` 
`make all -j128 --environment-overrides VLEN=128 XLEN=32   `

ksco, 2024-12-18:
It seems XLEN=32 was not well tested in some cases, I'll check later.

chenyuanyping, 2024-12-19:
Thanks,I have another question, I want to change the memory address of `pspike`, but it didn’t seem to take effect at the time. Where should I change it?
[image]
[image]

chenyuanyping, 2024-12-19:
normal spike can do  `spike -m0x0:0x400000`,But pspike doesn’t seem to work

ksco, 2024-12-20:
https://github.com/chipsalliance/riscv-vector-tests/commit/5bd2b3d7a43aa129d6d3e7f24bacf1f576ecbe3a should fix the XLEN=32 issue, please open a new ticket for other issues if any.

closed 2024-12-20 by ksco

chenyuanyping, 2024-12-23:
Thank you for fixing the issue!