chipsalliance/riscv-vector-tests #64: Generating tests for VLEN=64 fails

EAlexJ, 2025-02-08:
When building executables with VLEN equal to 64, faulty patches are generated for at least one instruction.
For XLEN=32:
```
build/merger -stage1output out/v64x32machine/tests/stage1/ -stage2output out/v64x32machine/tests/stage2/ -stage2patch out/v64x32machine/patches/stage2/
fatal: wrong patch generated for file vloxei16_v-0.S, expected 190, got 156
```
and for XLEN=64:
```
build/merger -stage1output out/v64x64machine/tests/stage1/ -stage2output out/v64x64machine/tests/stage2/ -stage2patch out/v64x64machine/patches/stage2/
fatal: wrong patch generated for file vloxei16_v-0.S, expected 191, got 181
```

ksco, 2025-02-08:
Thanks for the report, I'll take a look next week.

nadime15, 2025-07-30:
I’m not entirely sure, but this could be a Spike issue. In some cases, even when I set VLEN=64, Spike seemed to internally use VLEN=128.

I discovered this while running Spike in debug mode and printing VLEN and ELEN.

ksco, 2025-08-19:
So a change to the isa string in Makefile would resolve this issue?

nadime15, 2025-08-19:
For VLEN=64 and VLEN=32 (I think we can start supporting VLEN=32), yes. I will open a PR later.