chipsalliance/riscv-vector-tests #74: How to disable other lmul except lmul=1

iknowzxc, 2025-06-26:
I modified code in insn_util.go to this
`var allLMULs = []LMUL{1}
var wideningMULs = []LMUL{1}
var validLMULs = map[LMUL]struct{}{
	allLMULs[0]: {},
}`

But compilation has error
`2025/06/26 08:03:42 unreachable
2025/06/26 08:03:42 unreachable
make: *** [Makefile:121: generate-stage1] Error 1
make: *** Waiting for unfinished jobs....`

What should I do

iknowzxc, 2025-06-26:
> What is the use case? This does not make sense to me.

I want all generated cases to use LMUL = 1 only, and not allow values like 1/2, 1/4, 2, 4, or any others.

ksco, 2025-06-26:
Hmm okay, maybe try this?

```
var allLMULs = []LMUL{1}
var wideningMULs = []LMUL{1}
var validLMULs = map[LMUL]struct{}{
	LMUL(1) / 8: {},
	LMUL(1) / 4: {},
	LMUL(1) / 2: {},
	LMUL(1): {},
	LMUL(2): {},
	LMUL(4): {},
	LMUL(8): {}
}
```

But you do realize there is EMUL, which is not always 1 even if you restrict all the LMUL to 1, right?